DigitalShoestringSolutions/PowerMonitoring #14: Selection of basic ADC

tobyaharris, 2024-04-22:
How can we select ADC (eg BCRobotics or Grove etc) when building a basic solution?
The placeholder in line 18 of 80d81569498da2d54690abba861088e858beadac was never realised.

tobyaharris, 2024-04-22:
In the "nonuser" config file there is line which should do it. 
https://github.com/DigitalShoestringSolutions/PowerMonitoring/blob/4b6625a0f690af022bf046d64f5f9e161e9ff84a/current_sensing/config/pm_b_1p.toml#L20 
Can this be overwritten from user_config?

Greg-Hawkridge, 2024-05-08:
There has to be a compromise between something custom built for the solution (as in power monitoring basic) and a more general sensing service module. It could be as simple as changing the device module, but only if they both used the same interface. If we want to keep the user config simple, this would need to be addressed by implementing by having separte base config files for the basic power monitoring alternatives.

tobyaharris, 2024-06-12:
I'm looking forward to linking the Power Basic guide to this codebase instead of [it's own](https://github.com/DigitalShoestringSolutions/PowerMonitoringBasic/), but can I write explicitly that we can't move forward with this until PowerMonitoring has user-ready support for the minimum set of hardware. Do I remove the Grove ADC from the Basic guide? I would like to make a priority PR for this but I still haven't a clue how this sensing module works under the hood.

tobyaharris, 2024-07-05:
Adequately fixed in #31 , even if not perfectly slick.

closed 2024-07-05 by tobyaharris