DigitalShoestringSolutions/PowerMonitoring #17: Slow running on Pi 4B 2GB

tobyaharris, 2024-05-08:
Another test build in the lab - 

Everything is running very slowly. It takes several minutes to create a query in Influx's data explorer (`Loading tag values..` etc). Resubmitting the query takes 10-20s to fetch the data from the last 5 minutes (single phase basic). 
The Grafana dashboard took about 10 minutes to display the first data and only manages to update once every few minutes (autorefresh set to 5s)

This is the only solution installed on the pi. The SD card is 32GB. I'm using the Live Individual Simple dashboard, not even going near the Trend graphs. 

This repo is considerably computationally heavier than PowerMonitoringBasic. I would consider the 2GB RAM Pi4 unsupported. 
Is there anything we can do to lightweight this solution? It's creeping out of the remit of Minimal and Low Cost Hardware.

Greg-Hawkridge, 2024-06-24:
Yes

closed 2024-06-24 by Greg-Hawkridge

tobyaharris, 2024-06-24:
^ that is to say, we officially consider the 2gb to be unsupported hardware for use of this code and we should recommend the 4/8gb as standard.

tobyaharris, 2024-12-12:
Our advanced demonstrator still needs upgrading, maybe to a Pi5 8GB. It is very fustrating to use as is.