DigitalShoestringSolutions/PowerMonitoring #27: Support for n-phase machines

tobyaharris, 2024-05-30:
Whilst most machines are single phase or 3 phase, some are not. 

Some machines have multiple power supplies.
Some machines work intricately together in groups so that the whole cell can only be monitored as a single unit. 

If someone wants to use a shoestring solution to monitor what they call a machine with 2, 4, 6 or more phases, how can we support this?

A workaround is to consider the different supplies as separate machines and only look at the All dashboards, but once the deployment is scaled then they will need grouping.

tobyaharris, 2024-05-30:
For basic hardware, our ADCs have 8-16 inputs so this is only a code question. 

For intermediate hardware, could we reduce the many phases down to 3 by connecting multiple CTs in parallel? If the supplies are at all out of phase, the waveform might get confused. If both are fed from the same distribution board & phase then this should have the same effect as passing both conductors through the same CT. However, this does not feel like a good route to go down and I'd rather have multiple sensing systems and aggregate the data.

Greg-Hawkridge, 2024-06-11:
Shoestring aims to solve the typical use case. While there is some flexibility in the way solutions could be used, and end users could be creative about this - we shouldn't be designing solutions for edge cases. What are the 20% of features that address 80% of needs.

closed 2024-06-11 by Greg-Hawkridge

tobyaharris, 2024-06-12:
Can't argue with that. At least there's now a record here that we thought about it for future devs.