DigitalShoestringSolutions/PowerMonitoring #32: Smoothing control

tobyaharris, 2024-06-20:
I understand the data collection modules applies some smoothing to the data before publishing it to MQTT. 
I observe that after switching a load off (instantaneously), the reported values deplete in an S-curve profile over 15 seconds.

What if I want to turn the smoothing off and have more "raw" data reported? Is there a config file for this?

tobyaharris, 2024-06-20:
[image]

Greg-Hawkridge, 2024-06-20:
The only smoothing on the sensing side is that the basic power monitoring takes 5 samples each second and outputs the average - but that shouldn't lead to the situation above (assuming the sensed input is infact a step change). Intermediate and advanced record as they sense. There can be the appearance of smoothing due to aggregation on the display side of things - which may be what's seen here. Look at the raw data in the database and see if it agrees with the dashboard - if that is the case then that is what the sensor is reporting (reality seldom has actual step changes).

tobyaharris, 2024-06-20:
I believe it was a fast shutdown. I'll try again hard switching at the wall. 
Table view in Grafana shows exactly the same as the graph. One data point every 5 seconds. Steady state value, slightly below steady state, slightly above zero, steady zero.

tobyaharris, 2024-06-20:
Checking MQTT Explorer, samples are spaced every 1.1 seconds. This was a resistive load switched on and off at the wall. 
[image]
One non-steady state point I can understand, if it happened to sample right on the millisecond of inrush current etc. However on the switchoff there are clearly two points in between states - I don't believe it takes more than a second for the system to settle.

tobyaharris, 2024-06-20:
I suspect there is two issues here. One is with data capture, the second with aggregation and presentation. On the latter front, to go from the data I see in MQTT explorer to the table data in Grafana, there must be some kind of moving average going on. Above in MQTTE the steady state high and low points are 2-3s apart. How can Grafana have them 15s apart? How can there be more than one 5s group that is not steady state?

tobyaharris, 2024-06-20:
On the former front, here's as raw data as I can find (aggregatewindow removed): 
[image]

tobyaharris, 2024-06-24:
Next step - get the scopes out and understand the analogue electronics properly. Is the AC-DC amp inserting capacitance?

tobyaharris, 2024-08-12:
There's definitely some slow ramping going on. Here's the advanced dashboard a few seconds after switching our favourite fan heater on.
[image]
The voltages eventually (hard to say how long because it takes 30s+ to refresh (#17 )) settle at 240v.