DigitalShoestringSolutions/PowerMonitoring #34: Integration jitter

tobyaharris, 2024-06-26:
The integration is looking a bit unsteady. I tried shrinking the Time Buckets to enhance the issue and sure enough, buckets smaller than one minute make for very jittery Energy integrations, even when the power is steady:

[image]

With a time bucket approaching the order of magnitude of the sample rate, I wouldn't be surprised at some counting jitter due to significant relative variation in how many samples fall in each bucket.

I suspect the integrator is vulnerable to this because it might be summing each (sample value x time until next sample). How can we handle the time window edge cases? 

Although related I feel this is a different problem to #7 .

Greg-Hawkridge, 2024-07-01:
That will always be an issue - we should limit time bucket to being >10m

tobyaharris, 2024-07-05:
Closing as accepting the status quo, refinement effort not worth it. Time bucket default options are already suitable.

closed 2024-07-05 by tobyaharris