DigitalShoestringSolutions/PowerMonitoring #41: How to change sample rate?

tobyaharris, 2024-10-02:
Some solution builds in the field have a very poor (<50%) level of data collection uptime. Grafana shows regular large gaps in the timeseries.

Rumour has it that changing the sample rate may help the issue. Do we have any test results that support this?

Anyway, do we have any documentation on how to change the sample rate?
There's no `[sampling]` section of the config files like the old dc modules used to have. From looking I might guess `measurement.config.period`? Can this be done from the user config? What would an example `user_config.toml` look like that had a sample rate of 1 sample every 5 seconds?

Greg-Hawkridge, 2024-10-07:
measurement.config.period = 2.0

tobyaharris, 2024-10-07:
I'll ignore the 2 but take that as confirmation I was right about `measurement.config.period` as seen in https://github.com/DigitalShoestringSolutions/PowerMonitoring/blob/9451d4b746c922ace0d7ad28df9a5a84f6f9c0e0/current_sensing/config/pm_a_3p.toml#L47-L48
I'll add
```
[measurement.config] 
     period = 5.0 # seconds 
```
to the end of `user_config.toml`.

closed 2024-10-07 by tobyaharris