DigitalShoestringSolutions/PowerMonitoring #53: Current spikes of 242A and energy windows not plotted

tobyaharris, 2024-11-18:
Just checked the dashboard of our demonstrator for advanced power monitoring 3 phase this morning. 
The clamps are 60A/5A CTs and are configured as such on the MFM. 
The last 6 hours look like this:
[image]
Besides the test  between 11:40 and 11:56, all non-zero data points appear to be spurious noise. 
Zooming into the first of those green spikes at 0751:
[image]
It's a single reading of an impossible level of current.

These usually follow a data dropout. Perhaps caused by data collection restarting?
The other spikes peak to different values (122A / 120A also observed)

The engery spike becomes a more sensible box if the time bucket is reduced:
[image]
but why is the box of the same height not drawn at the 1hr bucket size? Zoom in on it as it is and the spike dissapears.

Greg-Hawkridge, 2024-11-18:
What do the log files say for that period? is that the value that was reported by the MFM or has something happened post sample?

tobyaharris, 2024-11-18:
<details><summary>current_sensing logs at 07:51</summary>

```
...
2024-11-18T07:51:13.646509+00:00 AdvancedPowerPi docker-current-sensing[1060]: ERROR:core.interface_modules.modbus:ERROR: exception in pymodbus Modbus Error: [Input/Output] ERROR: No response received after 3 retries
2024-11-18T07:51:13.650151+00:00 AdvancedPowerPi docker-current-sensing[1060]: ERROR:core.device_modules.multi_function_meter_HOBUT:Traceback (most recent call last):
2024-11-18T07:51:13.650478+00:00 AdvancedPowerPi docker-current-sensing[1060]:   File "/app/core/device_modules/multi_function_meter_HOBUT.py", line 87, in sample
2024-11-18T07:51:13.650922+00:00 AdvancedPowerPi docker-current-sensing[1060]:     readings[self.varI1] = await self.read_modbus_register(self.regI1)
2024-11-18T07:51:13.651151+00:00 AdvancedPowerPi docker-current-sensing[1060]:   File "/app/core/device_modules/multi_function_meter_HOBUT.py", line 135, in read_modbus_register
2024-11-18T07:51:13.651379+00:00 AdvancedPowerPi docker-current-sensing[1060]:     return await result
2024-11-18T07:51:13.651707+00:00 AdvancedPowerPi docker-current-sensing[1060]:   File "/app/core/interface_modules/modbus.py", line 61, in read_register
2024-11-18T07:51:13.652046+00:00 AdvancedPowerPi docker-current-sensing[1060]:     raise exc
2024-11-18T07:51:13.652353+00:00 AdvancedPowerPi docker-current-sensing[1060]:   File "/app/core/interface_modules/modbus.py", line 57, in read_register
2024-11-18T07:51:13.652599+00:00 AdvancedPowerPi docker-current-sensing[1060]:     modbus_response = await self.modbus_client.read_input_registers(address=register_addr, count=int(count),
2024-11-18T07:51:13.652806+00:00 AdvancedPowerPi docker-current-sensing[1060]:   File "/usr/local/lib/python3.9/site-packages/pymodbus/client/base.py", line 207, in async_execute
2024-11-18T07:51:13.653025+00:00 AdvancedPowerPi docker-current-sensing[1060]:     raise ModbusIOException(
2024-11-18T07:51:13.653364+00:00 AdvancedPowerPi docker-current-sensing[1060]: pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] ERROR: No response received after 3 retries
2024-11-18T07:51:13.653676+00:00 AdvancedPowerPi docker-current-sensing[1060]: ERROR:core.sensing_stack:Error during sampling: Traceback (most recent call last):
2024-11-18T07:51:13.654101+00:00 AdvancedPowerPi docker-current-sensing[1060]:   File "/app/core/sensing_stack.py", line 28, in execute
2024-11-18T07:51:13.654434+00:00 AdvancedPowerPi docker-current-sensing[1060]:     sample_dict = await sample_resp
2024-11-18T07:51:13.654759+00:00 AdvancedPowerPi docker-current-sensing[1060]:   File "/app/core/device_modules/multi_function_meter_HOBUT.py", line 128, in sample
2024-11-18T07:51:13.655064+00:00 AdvancedPowerPi docker-current-sensing[1060]:     raise e
2024-11-18T07:51:13.655304+00:00 AdvancedPowerPi docker-current-sensing[1060]:   File "/app/core/device_modules/multi_function_meter_HOBUT.py", line 87, in sample
2024-11-18T07:51:13.655508+00:00 AdvancedPowerPi docker-current-sensing[1060]:     readings[self.varI1] = await self.read_modbus_register(self.regI1)
2024-11-18T07:51:13.655692+00:00 AdvancedPowerPi docker-current-sensing[1060]:   File "/app/core/device_modules/multi_function_meter_HOBUT.py", line 135, in read_modbus_register
2024-11-18T07:51:13.656071+00:00 AdvancedPowerPi docker-current-sensing[1060]:     return await result
2024-11-18T07:51:13.656335+00:00 AdvancedPowerPi docker-current-sensing[1060]:   File "/app/core/interface_modules/modbus.py", line 61, in read_register
2024-11-18T07:51:13.656584+00:00 AdvancedPowerPi docker-current-sensing[1060]:     raise exc
2024-11-18T07:51:13.656852+00:00 AdvancedPowerPi docker-current-sensing[1060]:   File "/app/core/interface_modules/modbus.py", line 57, in read_register
2024-11-18T07:51:13.657386+00:00 AdvancedPowerPi docker-current-sensing[1060]:     modbus_response = await self.modbus_client.read_input_registers(address=register_addr, count=int(count),
2024-11-18T07:51:13.657696+00:00 AdvancedPowerPi docker-current-sensing[1060]:   File "/usr/local/lib/python3.9/site-packages/pymodbus/client/base.py", line 207, in async_execute
2024-11-18T07:51:13.658218+00:00 AdvancedPowerPi docker-current-sensing[1060]:     raise ModbusIOException(
2024-11-18T07:51:13.658605+00:00 AdvancedPowerPi docker-current-sensing[1060]: pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] ERROR: No response received after 3 retries
2024-11-18T07:51:13.658914+00:00 AdvancedPowerPi docker-current-sensing[1060]: ERROR:main.measure:Sample Error for device mfm1: Modbus Error: [Input/Output] ERROR: No response received after 3 retries - pausing for 10 seconds
2024-11-18T07:51:13.659257+00:00 AdvancedPowerPi docker-current-sensing[1060]: INFO:main.mqtt_out:pub topic:error/sensing_dc msg:{'type': 'device', 'id': 'mfm1', 'reason': 'Modbus Error: [Input/Output] ERROR: No response received after 3 retries', 'timestamp': '2024-11-18T07:51:13.648585+00:00'}
2024-11-18T07:51:24.073664+00:00 AdvancedPowerPi docker-current-sensing[1060]: WARNING:core.measurement_modules.gen_sample:Negative delay - resetting to period
2024-11-18T07:51:24.546696+00:00 AdvancedPowerPi docker-current-sensing[1060]: INFO:main.mqtt_out:pub topic:power_monitoring/Machine_1 msg:{'timestamp': '2024-11-18T07:51:24.072123+00:00', 'machine': 'Machine_1', 'power_real': 0.0, 'power_reactive': 0.0, 'power_apparent': 0.0, 'frequency': 49.94140625, 'power_factor': 0.0}
[... cut ...]
2024-11-18T07:51:26.410954+00:00 AdvancedPowerPi docker-current-sensing[1060]: INFO:main.mqtt_out:pub topic:power_monitoring/Machine_1/B msg:{'machine': 'Machine_1', 'timestamp': '2024-11-18T07:51:25.961953+00:00', 'current': 0.0, 'voltage': 241.6833038330078, 'phase': 'B'}
2024-11-18T07:51:26.412267+00:00 AdvancedPowerPi docker-current-sensing[1060]: INFO:main.mqtt_out:pub topic:power_monitoring/Machine_1/C msg:{'machine': 'Machine_1', 'timestamp': '2024-11-18T07:51:25.961953+00:00', 'current': 0.0, 'voltage': 241.7615966796875, 'phase': 'C'}
...
```

</details>
do show a reading of 241.73A just on phase A, but with no power.

This is on the first sample after recovering from #43 .

Greg-Hawkridge, 2024-11-25:
It looks like the modbus communication might be hitting the wrong register

tobyaharris, 2024-11-25:
Consider swapping to a USB to RS485 adapter? It was that way originally. Why did we switch away? Because the COM port numbers kept changing etc. Try all options again.

tobyaharris, 2024-12-11:
Every time I check our advanced power demonstrator, it is showing such spikes in the last few hours. I've lowered the logging level to try and see what registers are being hit (INFO logs above don't say much). Now waiting for the test setup to exhibit this while recording debug level logs. 
<details><summary>More INFO current sensing logs</summary>

```
2024-12-11T08:58:49.740712+00:00 AdvancedPowerPi docker-current-sensing[1171]: ERROR:core.sensing_stack:Error during sampling: Traceback (most recent call last):
2024-12-11T08:58:49.740950+00:00 AdvancedPowerPi docker-current-sensing[1171]:   File "/app/core/sensing_stack.py", line 28, in execute
2024-12-11T08:58:49.741214+00:00 AdvancedPowerPi docker-current-sensing[1171]:     sample_dict = await sample_resp
2024-12-11T08:58:49.741537+00:00 AdvancedPowerPi docker-current-sensing[1171]:   File "/app/core/device_modules/multi_function_meter_HOBUT.py", line 128, in sample
2024-12-11T08:58:49.741849+00:00 AdvancedPowerPi docker-current-sensing[1171]:     raise e
2024-12-11T08:58:49.742160+00:00 AdvancedPowerPi docker-current-sensing[1171]:   File "/app/core/device_modules/multi_function_meter_HOBUT.py", line 87, in sample
2024-12-11T08:58:49.742374+00:00 AdvancedPowerPi docker-current-sensing[1171]:     readings[self.varI1] = await self.read_modbus_register(self.regI1)
2024-12-11T08:58:49.742564+00:00 AdvancedPowerPi docker-current-sensing[1171]:   File "/app/core/device_modules/multi_function_meter_HOBUT.py", line 135, in read_modbus_register
2024-12-11T08:58:49.742767+00:00 AdvancedPowerPi docker-current-sensing[1171]:     return await result
2024-12-11T08:58:49.743043+00:00 AdvancedPowerPi docker-current-sensing[1171]:   File "/app/core/interface_modules/modbus.py", line 61, in read_register
2024-12-11T08:58:49.743311+00:00 AdvancedPowerPi docker-current-sensing[1171]:     raise exc
2024-12-11T08:58:49.743594+00:00 AdvancedPowerPi docker-current-sensing[1171]:   File "/app/core/interface_modules/modbus.py", line 57, in read_register
2024-12-11T08:58:49.743880+00:00 AdvancedPowerPi docker-current-sensing[1171]:     modbus_response = await self.modbus_client.read_input_registers(address=register_addr, count=int(count),
2024-12-11T08:58:49.744222+00:00 AdvancedPowerPi docker-current-sensing[1171]:   File "/usr/local/lib/python3.9/site-packages/pymodbus/client/base.py", line 207, in async_execute
2024-12-11T08:58:49.744561+00:00 AdvancedPowerPi docker-current-sensing[1171]:     raise ModbusIOException(
2024-12-11T08:58:49.744777+00:00 AdvancedPowerPi docker-current-sensing[1171]: pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] ERROR: No response received after 3 retries
2024-12-11T08:58:49.744964+00:00 AdvancedPowerPi docker-current-sensing[1171]: ERROR:main.measure:Sample Error for device mfm1: Modbus Error: [Input/Output] ERROR: No response received after 3 retries - pausing for 10 seconds
2024-12-11T08:58:49.745196+00:00 AdvancedPowerPi docker-current-sensing[1171]: INFO:main.mqtt_out:pub topic:error/sensing_dc msg:{'type': 'device', 'id': 'mfm1', 'reason': 'Modbus Error: [Input/Output] ERROR: No response received after 3 retries', 'timestamp': '2024-12-11T08:58:49.734093+00:00'}
2024-12-11T08:59:09.178789+00:00 AdvancedPowerPi docker-current-sensing[1171]: WARNING:core.measurement_modules.gen_sample:Negative delay - resetting to period
2024-12-11T08:59:09.663606+00:00 AdvancedPowerPi docker-current-sensing[1171]: INFO:main.mqtt_out:pub topic:power_monitoring/Machine_1 msg:{'timestamp': '2024-12-11T08:59:09.176492+00:00', 'machine': 'Machine_1', 'power_real': 0.0, 'power_reactive': 0.0, 'power_apparent': 0.0, 'frequency': 50.10968017578125, 'power_factor': 0.0}
2024-12-11T08:59:09.664841+00:00 AdvancedPowerPi docker-current-sensing[1171]: INFO:main.mqtt_out:pub topic:power_monitoring/Machine_1/A msg:{'machine': 'Machine_1', 'timestamp': '2024-12-11T08:59:09.176492+00:00', 'current': 240.79771423339844, 'voltage': 241.59385681152344, 'phase': 'A'}
2024-12-11T08:59:09.666101+00:00 AdvancedPowerPi docker-current-sensing[1171]: INFO:main.mqtt_out:pub topic:power_monitoring/Machine_1/B msg:{'machine': 'Machine_1', 'timestamp': '2024-12-11T08:59:09.176492+00:00', 'current': 0.0, 'voltage': 241.4206085205078, 'phase': 'B'}
2024-12-11T08:59:09.666976+00:00 AdvancedPowerPi docker-current-sensing[1171]: INFO:main.mqtt_out:pub topic:power_monitoring/Machine_1/C msg:{'machine': 'Machine_1', 'timestamp': '2024-12-11T08:59:09.176492+00:00', 'current': 0.0, 'voltage': 241.4837646484375, 'phase': 'C'}
2024-12-11T08:59:11.469887+00:00 AdvancedPowerPi docker-current-sensing[1171]: INFO:main.mqtt_out:pub topic:power_monitoring/Machine_1 msg:{'timestamp': '2024-12-11T08:59:11.059008+00:00', 'machine': 'Machine_1', 'power_real': 0.0, 'power_reactive': 0.0, 'power_apparent': 0.0, 'frequency': 50.11174392700195, 'power_factor': 0.0}
2024-12-11T08:59:11.470921+00:00 AdvancedPowerPi docker-current-sensing[1171]: INFO:main.mqtt_out:pub topic:power_monitoring/Machine_1/A msg:{'machine': 'Machine_1', 'timestamp': '2024-12-11T08:59:11.059008+00:00', 'current': 0.0, 'voltage': 241.5242919921875, 'phase': 'A'}
2024-12-11T08:59:11.472072+00:00 AdvancedPowerPi docker-current-sensing[1171]: INFO:main.mqtt_out:pub topic:power_monitoring/Machine_1/B msg:{'machine': 'Machine_1', 'timestamp': '2024-12-11T08:59:11.059008+00:00', 'current': 0.0, 'voltage': 241.21484375, 'phase': 'B'}
[... cut ...] '2024-12-11T08:59:11.059008+00:00', 'current': 0.0, 'voltage': 241.3248291015625, 'phase': 'C'}
```

</details>

Also, it's not always on phase A, nor always in the 240s, so there might be more going on here that just hitting a voltage register by accident. Here are all the non-zero current readings in the last 24h, 
[image]
^ agreggate window / mean filtering is of course removed from this query. The 50ish readings could be explained by hitting  a frequency reg, but I have no idea why the 2.1A.