DigitalShoestringSolutions/PowerMonitoring #58: WARNING: core.measurement_modules.gen_sample:Negative delay - resetting adjustment

tobyaharris, 2024-12-20:
Warning on every sample cycle. 
(separately I'm concerened about all the readings being exactly 0)

[image]

It is not immediately clear which of these almost identical code blocks this is coming from:

https://github.com/DigitalShoestringSolutions/PowerMonitoring/blob/df38c70431d566ed3eb3d1e1ffcee1d624f1db23/current_sensing/code/core/measurement_modules/gen_sample.py#L75-L102

https://github.com/DigitalShoestringSolutions/PowerMonitoring/blob/df38c70431d566ed3eb3d1e1ffcee1d624f1db23/current_sensing/code/core/measurement_modules/gen_sample.py#L189-L215

https://github.com/DigitalShoestringSolutions/PowerMonitoring/blob/df38c70431d566ed3eb3d1e1ffcee1d624f1db23/current_sensing/code/core/measurement_modules/gen_sample.py#L314-L339

PS looking at the repetition of code in these classes, would this benefit from using a bass / parent class and deriving the minor variations?

Greg-Hawkridge, 2025-01-10:
What this warning indicates is that the solution is taking so long to sample the data source that it can't achieve the prescribed sample rate (in other words it needed to have started sampling in the past to make the next target sample timestamp) - so the solution is giving up on maintaining an optimised delay that accounts for sample time and is just starting again.

Given this in on an intermediate or advanced I wouldn't be surprised if there is some sort of network level issue that is timing out - although unexpected that the response is a zero reading.

The specific code that is throwing this error will depend on which measurement class is specified in the config file.

tobyaharris, 2025-01-10:
This was Basic! Offline I can give you more details about which project / customer.

tobyaharris, 2025-01-10:
My suspicion is that the warning is inevitable as way the delay might be calculated can't allow for 15 sps? 

the +1 in this line offsets all the samples? 
https://github.com/DigitalShoestringSolutions/PowerMonitoring/blob/df38c70431d566ed3eb3d1e1ffcee1d624f1db23/current_sensing/code/core/measurement_modules/gen_sample.py#L82

There's no reason a 75ksps+ adc shouldn't confidently deliver 15sps.

tobyaharris, 2025-01-13:
Try to reproduce. Do a BCR 3pu build in lab. See if debug logs can provide more timestamp details about where the delay could be coming from. There's no obvious place in the code that should be slow.

tobyaharris, 2025-03-13:
Still seeing this even with the simulator 

[image]

tobyaharris, 2025-09-29:
Moved to https://github.com/DigitalShoestringSolutions/sm_SensingDC/issues/13

closed 2025-09-29 by tobyaharris