DigitalShoestringSolutions/PowerMonitoring #59: Historic (All) dashboard joins points between gaps in data

tobyaharris, 2025-01-13:
[image]
I believe it's our general policy (undocumented?) to only join over gaps <1minute long

Examples where that has been previously implemented:
https://github.com/DigitalShoestringSolutions/PowerMonitoring/issues/7#issue-2222390673
https://github.com/DigitalShoestringSolutions/AirQualityMonitoring/pull/8
https://github.com/DigitalShoestringSolutions/AirParticleMonitoring/pull/5

I propose this dashboard is updated with similar changes.

Greg-Hawkridge, 2025-02-03:
Go for it

tobyaharris, 2025-02-03:
It's all of them.
[image]
As I'm not certain about if `insertNulls` needing to be removed, I'll do this in Grafana.

tobyaharris, 2025-03-24:
Github workflows are not very good at automatically closing issues affter their intermediary release branch has been merged into main...

PR #65 (2025-03-20): Dashboard consistency
Minor edits to all dashboards to make them more consistent.

* All dashboard json files end with one blank line
* Live dashboards have a time range of Last 5 Minutes, Historic 6 hours and Trend 7 Days.
* Live dashboards refresh every 5s, Historic and Trend do not autorefresh 
* `createEmpty: true` and `Connect Null Values: Threshold <1m` where appropriate (fixes #59 )
* Removed `Machine_1` etc from dashboard json files (machine variable is None)
* Voltage display renamed to Measured Voltage (for consistency with power factor)
closed 2025-03-24 by tobyaharris