DigitalShoestringSolutions/PowerMonitoring #69: Negative power factor and real/reactive power

tobyaharris, 2025-04-24:
Here is a quick screenshot of a customer deployment, advanced 3pu configuration. 

[image]

Firstly there is an issue that the currents are unreasonably large, I think a decimal place was slipped on the MFM config. Ignoring that, there remains a polarity issue.

The power factor, real power and reactive power are all negative. The monitored device is a load. Could this be due to connecting one or all of the clamps the wrong way round? Yet the reported currents are still positive. Could the MFM be quietly ignoring the sign of current in reporting, but considering it a negative value when calculating powers? 

Could this be due to the phase order being mismatched between the voltage and current monitoring? 

A true 3ph setup (120deg separation) would be helpful to test this. What we have in the lab is only a single phase split between different loads.

tobyaharris, 2025-04-24:
Possibly relevant, here is the historic dashboard

[image]

tobyaharris, 2025-04-28:
Greg: Check other MFM settings. 3p3u/3pu?

Greg-Hawkridge, 2025-07-28:
@tobyaharris check if issue still present after redoing settings

Greg-Hawkridge, 2025-09-29:
Company has fixed the issue - @tobyaharris to find out what the solution was